vtyulb/BSA-Analytics #18: packet precise search processing

vtyulb, 2015-04-01:


STyulb, 2015-04-01:
Пока не сделанные задачи
1) обработка пульсара на краю файла. Какой алгоритм в программе?
2) пакетный режим обработки
3) построение динамических спектров
4) определение плотности потока известных пульсаров
5) основы тайминга (улучшение точности периода не хуже пятого знака) для многократно повторяющихся в разные дни пульсаров.
6) определение координаты пульсара по прямому восхождению, записываемой в каталог. Какой алгоритм в программе? Здесь же: Фиксация фазы и точного времени для пульсара.
7) определение дисперсии. Какой алгоритм в программе?

STyulb, 2015-04-05:
Пакетный режим: открыл файл в котором содержаться данные для проверки обнаруженных пульсаров (время, дата, период, мера дисперсии, модуль, луч - таких строк может быть много); открыть файл для обработки; прочитать московское время начала записи и перевести в звездное время; прибавить один час, чтобы знать границы поиска по времени пульсара; провести поиск пульсара по данным заданным в строке. 

Планируется, что будут даны десятки (если не сотни и тысячи) кандидатов и сутки (или месяцы) исходных наблюдений.

closed 2015-04-07 by vtyulb

STyulb, 2015-04-09:
Славка, привет!
Задание на github я разместил. Здесь прикрепляю файл с левыми периодами.
        Пока,
               Папа
P.S. Да, за сегодня найден еще один пульсар!!!

07.04.2015 22:00, Vladislav Tyulbashev пишет:

> Closed #18.
>
> ---
> https://github.com/vtyulb/BSA-Analytics/issues/18#event-275584401

0.5060
0.5205
0.5315
0.5640
0.5680
0.5785
0.6120
0.6230
0.6400
0.6620
0.6860
0.6910
0.7340
0.7705
0.7995
0.8130
0.8660
0.8740
1.0000