vtyulb/BSA-Analytics #20: определение плотности потока известных пульсаров

vtyulb, 2015-04-01:


STyulb, 2015-07-24:
Режим калибровки. 
Две задачи: 
а) определение интегральной и пиковой плотности потока;
б) поиск гигантских импульсов.
 
1) Задаем время старта (соответствует координате пульсара по прямому восхождению на 2000 год и от него отступаем в каждую сторону примерно по 1.5 минуты), период пульсара, меру дисперсии, имя файла, модуль, луч.
2) Ищем ближайшие ступеньки к пульсару (между которыми он находится). Если одной из ступенек нет в данных (не хватает файла), то вся работа производится по одной ступеньке. Полная длительность ступеньки - это стандарт и составляет примерно 15 секунд и примерно равные доли по 5 секунд: первый минимум, максимум, второй минимум. Начиная с 2013 года ступеньки запускаются в одно и то же московское время и могут быть найдены в файлах: дата_01(05,09,13,17,21)_N1/N2_00.pnt(pnthr). Номера точек, которые нужно будет усреднять от 3030 до 3050 (первый минимум), от 3095 до 3105 (максимум ступеньки), от 3135 до 3155 (второй минимум). Предусмотреть галочку (по умолчанию опция отключена) для ручного ввода координат ступеньки. Ступенька в очень редких случаях может быть пропущена. Если есть лишь одна ступенька, то все дальнейшие процедуры делать по ней. Поэтому нужно потестировать такой случай для автоматической обработки наблюдений! 
	Здесь расписан случай коротких данных. Для длинных данных все номера точек умножаются на коэффициент постоянная_времени_коротких_данных/пост._врем._длин._данных и округляются до ближайшего целого числа.
3) Определяем высоту ступенек (h_i, i - номер частотного канала) в каждом частотном канале и в общем канале. Т.е. проводим усреднение по двум минимумам (в сумме по 40 точкам), в указанных в пункте (2) границам, так же делаем усреднение по 20 точкам в максимуме ступеньки и вычисляем разницу между максимумом и минимумом. Для каждого частотного канала и для общего канала вычисляем коэффициенты 2400/h_i (высота ступеньки в единицах АЦП).
4) Делаем компенсацию плотности потока (нормировку) в каждом частотном канале и в общем канале. Для этого все точки внутри заданного временного интервала в каждом частотном канале домножаем на среднее между полученными в пункте (3) коэффициентами (ВСТАВКА). Затем, по заданным в пункте (1) данным, получаем средний (одинарный) профиль пульсара по отнормированным данным. (По умолчанию чистку шумов не делать. Галочка "чистка шумов" остается как и в точном поиске). Если средняя линия проведена правильно, то есть относительно этой линии шумы выше и ниже нее распределены одинаково, то можно переходить к пункту (5). Если на экране видно, что нулевой уровень проведен не правильно, то нужно сделать компенсацию нулевого уровня. Для этого предусмотреть очередное окошко "нулевой уровень" в котором по умолчанию стоит число "0". Т.к. по вертикальной оси отложены истинная высота среднего профиля в каких-то нормированных единицах АЦП, то я могу вместо нуля поставить в это окошко положительное или отрицательное число. Если число положительное, то во всех дальнейших вычислениях нужно будет прибавлять это число к каждой точке массива, выведенного на экране. Если число отрицательное, то нужно будет вычитать это число из каждой точки массива.
5) а) Складываем значение всех точек массива в котором содержится средний профиль. Делим на количество точек. Получившееся значение называется интегральная плотность потока в Кельвинах (S_int) и запоминается. Возможно лучшим вариантом окажется, если можно будет указать границы суммирования в номерах точек. Если будет суммироваться лишь импульс, то неправильно определенный нулевой уровень будет давать меньший вклад в погрешность. В любом случае какое-бы суммирование не было, нужно делить на общее количество точек в полном среднем профиле.
      б) в отдельном окошке должна быть возможность записать число означающее полуширину среднего профиля в точках и долях точки. Определяем пиковое значение плотности потока как S_peak=S_int * P(период в точках и долях точки)/W_e(полуширина среднего профиля в точках и долях точки). 
6) Программа по поиску гигантских импульсов пульсаров (будет делаться исключительно(!!!) по длинным данным). До части ВСТАВКА в пункте (4) все так же. Компенсируем дисперсию и складываем все импульсы с учетом DM. Получаем трехминутный массив в котором друг за дружкой идут импульсы. Вычисляем сигму шумов в этом файле на 20 секундных интервалах. Выбираем сигму шумов медианное из 9 посчитанных сигм. 
         Далее есть два разных определения гигантского импульса. Во-первых, если у нас посчиталась интегральная плотность потока (для тех пульсаров, которые мы можем обнаружить в наших наблюдениях), тогда гигантский импульс - это одиночный импульс пиковое значение которого в 10 и более раз больше, чем найденное нами интегральное значение плотности потока пульсаров. В выходном файле нужно записывать звездное время этого импульса до долей секунды (что-бы его можно было посмотреть в первичных записях и однозначно отождествить), интегральную плотность потока, сигма шумов, высоту и частное (высота единичного импульса деленная на интегральную плотность потока). Во-вторых, если интегральная плотность потока не обнаруживается (т.е. пульсар сам по себе слабый, но имеет гигантские импульсы), то сравнивается сигма шумов с любыми пичками в записи и проверяется условие, что пичок в 4 раза больше сигма шумов. Нужно проверить при этом, чтобы при заданной нами ранее мере дисперсии SNR этого пичка был максимальный (иначе это помеха).  Если условие соблюдается, то в файл записывается примерно то же, что и в пункте "во-перв [... cut ...] диничного импульса, деленной на сигму шумов.
7) При включении режима калибровки необходимо, чтобы динамический спектр, если мы его будем разрисовывать, использовал нормированные значения всех точек. Если нажата галочка "Поиск гигантских импульсов", то на динамическом спектре должен разрисовываться один импульс, координаты которого по звездному времени указаны (по нижней оси так будет один (вместо двух как у нас сейсчас) период пульсара, а по верхней частоты, на которых наблюдается этот импульс).

vtyulb, 2015-08-08:
flow done

vtyulb, 2015-08-10:
Вместо супер изменений в спектре, нужно всего лишь запустить его построение с очень большим периодом (например, периодом в 200 секунд)

closed 2015-08-10 by vtyulb

STyulb, 2015-08-20:
http://cont.ws/post/112514