vtyulb/BSA-Analytics #22: поиск дупликатов

STyulb, 2015-04-05:
В программе поиска дупликатов (и, естественно, в analytics) нужно ужесточить условия подтверждения того, что дупликат найден. 
1) Оставить в ходе поиска проверку лишь для половинных периодов, двойных и тройных. За основной период принимается период введенный в специальном файле или в поисковом активном окошке.
2) Условие подтверждения дупликата для коротких данных следующее:
модуль(дробная часть от [кратный период/главный период] * целая часть от [длина записи (200сек)/главный период])<0.1*Period_Pulsara - дупликат найден!!!
если 
модуль(дробная часть от [кратный период/главный период] * целая часть от [длина записи (200сек)/главный период])< 1.5*постоянная времени, то использовать именно это условие

closed 2015-04-05 by vtyulb

STyulb, 2015-04-09:
Слава, привет!
Ты слишком рано сделал архив за 020914, т.к. там был сбой и не посчитан 
час 00. Его придется заново запускать и считать.
       Пока,
             Папа

05.04.2015 18:22, Vladislav Tyulbashev пишет:

vtyulb, 2015-04-09:
Так я не могу его посчитать, так как падает на нем все. Отложен он пока

STyulb, 2015-04-10:
o'k
Тогда я перекачиваю этот архив. Обойдемся без его счета.
     Пока,
              Папа
P.S. Спасибо за кнопку "левые" пульсары. Пока ничего не проверял. 
Сегодня надеюсь закончить поиск пульсаров с SNR>6 и установлю последнюю 
сборку.

09.04.2015 20:14, Vladislav Tyulbashev пишет:

vtyulb, 2015-04-10:
Я его удалил уже...
Через пару часов создастся.