vtyulb/BSA-Analytics #65: программа поиска гигантских импульсов

STyulb, 2016-01-11:
1) Программа не работает с "длинными" данными под виндами 
2) табличка с номером точки, где находится гиг.импульс ложится поверх спектра (приходится все время ее сдвигать). Сделать, чтобы она была сбоку от спектра.
3) на табличке с номером гиг.импульса давать дополнительную информацию: а) сигма шумов на периоде пульсара; б) плотность потока, определенная по ступеньке.. Эта информация позволит сразу оценивать мощность импульса.

vtyulb, 2017-02-11:
Из задачи #67
> Забыл написать. На выводе помимо номера точки максимума импульса, сигмы шумов, плотности потока источника нужно также выводить значение высоты импульса (т.е. значение = значение в максимуме - значение в минимуме (а лучше минус значение в средней точке))

vtyulb, 2017-03-05:
Сделано

closed 2017-03-05 by vtyulb