vtyulb/BSA-Analytics #75: доработка pulsar fourier analytics

STyulb, 2016-08-07:
1) нет возможности загрузить много блоков
2) пульсары, которые в каталоге имеют белые строки (месячное усреднение), не имеют координаты по звездному времени
3) с учетом предыдущего БОЛЬШОГО задания, где я сделал определение медианного сигма. В каталог отобранных пульсаров должны входить ВСЕ объекты с SNR больше заданного. Т.е. если в каком-то файле видны несколько гармоник, то все они должны отображаться в каталоге, если отвечают условию SNR больше заданного значения
4) в правом нижнем углу есть текущая координата точки, которая соответствует положению мышки. Хорошо бы через точку с запятой выдавать там текущую частоту и текущий период.

vtyulb, 2016-08-07:
1) и в ближайшее время не будет
2) теперь имеют
3) in progress
4) done

vtyulb, 2016-08-14:
all done

closed 2016-08-14 by vtyulb