vtyulb/BSA-Analytics #92: последовательность выполнения задач

STyulb, 2016-10-08:
1) Работа с нижней площадкой. Учет того, что частоты и направления внизу расходятся. Это задачи 86 и 87 из списка
2) Не работает режим для динамического спектра, перебора периодов и мер дисперсий для длинных данных. На коротких данных динамический спектр выглядит странно (попробуй взять любой сильный известный пульсар и увидишь).
3) Нужна возможность корректного выбора лучших данных из посчитанных спектров. Колонка 4 при поиске пульсаров "Pulsar search fourie" использовалась для меры дисперсии. Теперь в непонятном для меня режиме для коротких данных там находятся нули и единицы взятые из неизвестного фильтра или большое число, которое указывает на отброшенные данные. Для длинных данных эта колонка вообще не работает. Там всегда стоит единица. Мне нужно, чтобы в этой колонке учитывались фурье фильтры, а именно: а) при нажатии "хорошо выглядящие спектры" для индивидуальных спектров стояла "1" если спектр хорошо выглядит и "большое число" если этот спектр отброшен из складывания; б) при дополнительном нажатии "выбрать лучшие" значение "1" должно остаться лишь для того количества лучших, что мы написали (лучшие - это файлы с наименьшей медианной дисперсией), для оставшихся должно стоять "0" или "большое число" для откинутых с самого начала; в) при дополнительном нажатии "peak from" и указании на каких SNR выбирать пики, значение "1" должно остаться для тех индивидуальных спектров у которых сигнал к шуму больше заданного значения. Пункты б) и в) могут работать совместно, а могут работать и независимо. На настоящий момент что-то работает для коротких данных, для длинных все эти фильтры не срабатывают. В любом случае все фильтры основаны на сигме шумов, а она считается не корректно.
4) Мне нужна возможность работы с индивидуальными спектрами такая же как раньше была для индивидуальных записей. То есть для данного блока должна быть таблица всех обнаруженных на заданном сигнал шуме периодических сигналов, а затем работа с этими сигналами с использованием верхней части  "Analytics". Для этой работы опять же нужно абсолютно корректное определение сигмы шумов.

vtyulb, 2016-10-11:
1,2 - done

vtyulb, 2016-10-12:
3 - done

vtyulb, 2016-10-17:
4 - добавить работающий фильтр SNR в серой зоне

vtyulb, 2016-10-20:
Вдохновение таки пишло

closed 2016-10-20 by vtyulb