vtyulb/BSA-Analytics #196: Поиск сильных транзиентов с малыми мерами дисперсии

STyulb, 2017-04-14:
Если мера дисперсии импульса от 0 и до 2 (может быть 3), то в динамическом спектре в первом приближении будет вертикальная прямая. Как нам ее отличить от помехи???
Предлагаю:
1) Работать лишь с 33 каналом
2) Достаем полученную нами информацию по лучшим записям, которые предназначены для суммирования
3) На этих записях (блоках) смотрим импульсы (помеховые пики), имеющие SNR>6.
4) Запоминаем количество импульсов в данном блоке по всем выбранным записям. 
5) Получаем статистику. Так как помехи в целом случайные, то при накоплении количества помех в блоке за все время должно получится гауссовское распределение. То есть будет некое среднее количество помех в блоках (мы эту оценку сделаем для всех блоков сразу) и будет небольшое количество блоков с очень малым количеством помех и с очень большим количеством помех. Нас будут интересовать случаи, когда в каком-то блоке по звездному времени пишется слишком много помех. Предполагая, что это отдельные импульсы близких пульсаров, будем их исследовать. В принципе, для сильных "помех" даже мера дисперсии 1 будет отслеживаться.
6) Выходная информация: блок, дата, модуль, луч, SNR "помехи" в любой форме. Думаю, что это подгоняется под нашу стандартную программу просмотра. Какие-то окошки будут не активными. В просмотровой части хорошо бы иметь дополнительно возможность вызова динамического спектра с тем, чтобы эту "помеху" в реальности.

vtyulb, 2017-04-23:
> К пункту 5 из задачи 196.
Для пиков, которые обнаруживаются не более чем в 2 лучах на SNR>6 вести статистику в отдельном файле. Этот файл должен изменяться (пополняться), когда проводится счет очередных данных. В файле должна быть следующая информация для каждого блока:

дата, когда найден пик
имя файла
модуль и луч
сколько пиков найдено в этом модуле и луче за эту дату
сколько дней просчитано (т.е. не выкинуто, а реально посчитано, пусть даже за данный день не найдено ни одного пика)
сколько пиков в найдено в данном модуле и луче за все посчитанные даты
При загрузке этого файла на экран должно выводится 48 точек (гистограмма). Значение в каждой точке - это количество найденных импульсов, удовлетворяющих поставленной задаче. При подводе курсора к точке должна появляться информация в любом удобном месте:

Количество импульсов
Модуль и луч
Количество просчитанных дней см. выше
номер блока

vtyulb, 2017-06-05:
Здесь уже в общем-то все сделано, поиск теперь будет идти от дисперсии = 2

closed 2017-06-05 by vtyulb