vtyulb/BSA-Analytics #198: Поиск сильных транзиентов с большими мерами дисперсии и одиночными всплесками.

STyulb, 2017-04-15:
Для пульсаров известно такое явление как рассеяние. Оно заключается в том, что если есть много межзвездной среды (большая мера дисперсии) между пульсаром и наблюдателем, то импульс рассеивается. Вместо дельта-функции с которой мы привыкли иметь дело получается импульс, который имеет меньшую высоту, его левый край резко падает как у ступеньки, а правый край падает по экспоненте от вершины. При этом общая площадь под импульсом равна площади под старой дельта функцией, но сигнал к шуму очень плох. При этом рассеяние на одной и той же мере дисперсии может отличаться в 10 раз и поэтому нельзя ввести какую-то стандартную форму и сравнивать получившиеся сигналы с ней. Т.е. при работе с рассеянными импульсами нужно иметь методику для их восстановления.
1) Берем хорошие отобранные записи.
2) Делаем попарную циклическую кросскорелляцию: первая частота (самая низкая) со второй, третья частота с четвертой, пятая с шестой и так далее. Сдвиг при кросскорелляции доводим до точки 2048. Сдвиг в массивах делаем следующим образом: более низкая частота в паре частот стоит на месте, более высокая сдвигается по ходу времени. (Возможно нужно будет извлечь корень изо всех точек полученного массива). Получаем 16 кросскорелляционных функций, складываем их и делим на 16.И
3) Ищем пики на CCF. Расстояние от первой точки до пика однозначно определяет DM. Расстояние между пиками определяет период. 
4) Метод хорошо работает для одиночных пиков и для пульсаров с большим периодом. На пульсарах с малым периодом будет сбоить.
5) Теоретически этот метод также сработает и на объектах с малой мерой дисперсии, но будет тоже самое, что для Фурье на низких частотах, когда из-за шумов ничего не разобрать
6) Сохранять то же, что и в предыдущих задачах по поиску всплесков.
7) Если принято решение, что объект нужно доииследовать, то нужно восстановить время начала импульса и для этого улучшить сигнал к шуму восстановлением формы импульса. Как это делать я еще подумаю

vtyulb, 2017-04-16:
Дополнение:

> Нужно делать не только кросскорелляцию не 1 с 2, 3 с 4 и т.д., а затем сложение 16 CCF (это хорошо для больших DM), но и кросскорелляцию 1 частотного канала с 17, 2го с 18, 16го с 32, а затем сложение. Второй вариант будет хорошо работать с малыми DM (от DM 3-4 и до DM 30-50).

vtyulb, 2017-07-21:
1 - проигнорировано, скорости хватает на обработку всего
2/дополнение - делаю кросс корреляцию всех каналов сразу
3 - DM сбрасывается, период можно будет вручную восстанавливать
4,5 - малые дисперсии вырезаются из обработки
6 - сделано

Запустил тестовый месяц на обработку

closed 2017-07-21 by vtyulb