vtyulb/BSA-Analytics #221: картинки по частотным каналам (напоминание)

STyulb, 2017-04-25:
Напоминаю о разговоре по поводу построения аналога динамических спектров.
Задается: Центральное время (координата транзиента); сколько секунд (минут) показывать по обе стороны от этой точки;
На экран выводится: Содержимое частотных каналов друг под другом с минимальным перекрывавом или без перекрыва. Полный аналог в статье Самодурова. 
а) сделать по вертикальной оси подпись на экране с частотами каналов. 
б) по горизонтальной оси отражать как обычно номера точек и текущие координаты по звездному времени. 
в) минимальное время отражаемое на экране +-5 секунд от центральной точки; максимальное время на экране +-150 секунд от центральной точки.
г) для печати нужны оба варианта (т.е. с подписями на экране и также имя файла, модуль, луч как ты уже делал. Второй вариант вывод только линии частотных каналов (нужно еще будет попробовать с частотами, но если они будут плохо видны на рисунках в печатные статьи, то смысла нет).

vtyulb, 2017-05-01:
Запихать в precise search модуль transient detalization

vtyulb, 2017-06-05:
Сделано

closed 2017-06-05 by vtyulb