vtyulb/BSA-Analytics #274: баг в картинке выдачи индивидуальных импульсов

STyulb, 2017-06-14:
16 каналов верхних, отличается от 16 каналов нижних. В рисунке виден явный излом. Рисунок прилагаю.
[image]

vtyulb, 2017-06-14:
Программа всегда работает одинаково. Из-за разного усиления по каналам нижние раздвинулись чуть сильнее, верхние чуть слабее. При нормализации этот эффект исчезнет. Могу всегда раздвигать каналы на одинаковую величину, тогда какие-то будут серьезно наползать на другие, либо же все будет слишком плоское

vtyulb, 2017-06-15:
Короче, надо посмотреть как будут выглядеть подобные импульсы в нормированном поиске. Потому что строгое расстояние между каналами даст просто другое искажение. Пока задачу закрываю

closed 2017-06-15 by vtyulb